micropython/micropython-lib #746: Only able to publish 1 message with umqtt.simple

Stubbs, 2023-10-10:
I'm trying to send config info to Home Assistant from a Pico W with the following code:

```
import time
import config
from network import WLAN,STA_IF
import ubinascii
# import gc
# import micropython

from machine import Pin, UART
from pimoroni import RGBLED, Button
# from pms5003 import PMS5003
from pimoroni_i2c import PimoroniI2C
from breakout_bme68x import BreakoutBME68X, STATUS_HEATER_STABLE
from breakout_ltr559 import BreakoutLTR559

from umqtt.simple import MQTTException, MQTTClient

# set up the LED
led = RGBLED(6, 7, 10, invert=True)
led.set_rgb(200,100,0)

# micropython.mem_info()
# Connect to the wifi
wlan = WLAN(STA_IF)
wlan.active(True)
wlan.connect(config.SSID, config.PASSWORD)
unique_id = ubinascii.hexlify(wlan.config('mac')).decode().upper()

# Connect to the MQTT Server
mqtt_connected = False;
try:
    mqtt = MQTTClient("enviro_plus", config.MQTT_SERVER, config.MQTT_PORT, config.MQTT_USER, config.MQTT_PASSWORD)
    mqtt.DEBUG = True
    mqtt.connect()
    mqtt_connected = True;
except MQTTException as e:
    print(f"Unable to connect to MQTT: {e}")
except OSError as e:
    print(f"Unable to connect to MQTT: {e}")

# Initialise Home Assistant.
if mqtt_connected:
    # print('Initial free: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_gas/config1", f'{{ "name":  "Gas resistance",   "state_topic": "homeassistant/sensor/enviroplus_{unique_id}/state",  "value_template": "{{{{ value_json.temperature }}}}",  "unique_id": "enviroplus_{unique_id}_temp", "unit_of_measurement": "Ω",  "device": {{"identifiers": ["enviroplus_{unique_id}"], "name": "Workshop Environment"}}}}   ')
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_pm1/config1", f'{{ "name": "PM1", "device_class": "pm1", "stat_t": "homeassistant/sensor/enviroplus_{unique_id}/state", "value_template": "{{{{ value_json.pm1 }}}}", "unique_id": "enviroplus_{unique_id}_pm1", "unit_of_measurement": "µg/m³","device": {{ "identifiers": "enviroplus_{unique_id}", "name": "Workshop Environment"}}}}   ')
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_pm25/config1", f'{{ "name": "PM2.5", "device_class": "pm25", "stat_t": "homeassistant/sensor/enviroplus_{unique_id}/state", "value_template": "{{{{ value_json.pm25 }}}}", "unique_id": "enviroplus_{unique_id}_pm25", "unit_of_measurement": "µg/m³","device": {{ "identifiers": "enviroplus_{unique_id}", "name": "Workshop Environment"}}}}   ')
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_pm10/config1", f'{{ "name":  "PM10",  "device_class": "pm10",  "state_topic": "homeassistant/sensor/enviroplus_{unique_id}/state",  "value_template": "{{{{ value_json.pm10 }}}}",  "unique_id": "enviroplus_{unique_id}_pm10", "unit_of_measurement": "µg/m³",  "device": {{"identifiers": ["enviroplus_{unique_id}"], "name": "Workshop Environment"}}}}   ')
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_T/config1", f'{{ "name":  "Temperature",  "device_class": "temperature",  "state_topic": "homeassistant/sensor/enviroplus_{unique_id}/state",  "value_template": "{{{{ value_json.temperature }}}}",  "unique_id": "enviroplus_{unique_id}_temperature", "unit_of_measurement": "°C",  "device": {{"identifiers": ["enviroplus_{unique_id}"], "name": "Workshop Environment"}}}}   ')
    mqtt.publish(f"homeassistant/sensor/enviroplus_{unique_id}_H/config1", f'{{ "name":  "Humidity",  "device_class": "humidity",  "state_topic": "homeassistant/sensor/enviroplus_{unique_id}/state",  "value_template": "{{{{ value_json.humidity }}}}",  "unique_id": "enviroplus_{unique_id}_humidity", "unit_of_measurement": "%",  "device": {{"identifiers": ["enviroplus_{unique_id}"], "name": "Workshop Environment"}}}}   ')
    mqtt.disconnect()
    # print('Initial free: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
else:
    print("MWTT not connected.")
```

There are a couple of issues, firstly I have to pad the paylod with 3 extra characters or the json that's sent in missing it's terminating brackets.

The other issue is that it only seems to send the first message, I have MQTT Explorer watching what's published and I only ever see the first of those published messages, no matter what order I put them in.

Stubbs, 2023-10-12:
A little more on this, from the mqtt logs themselves:

```
mqtt                        | 1697055632: New connection from 192.168.1.62:50051 on port 1883.
mqtt                        | 1697055632: New client connected from 192.168.1.62:50051 as enviro_plus (p2, c1, k0, u'ha').
mqtt                        | 1697055632: Client enviro_plus disconnected due to malformed packet.
```

Stubbs, 2023-10-13:
The issue is caused by "unit_of_measurement" where some of the values are UTF-8, even when I encode the strings with .encode() it still causes the protocol error.

The padding is a red herring.

hmaerki, 2023-10-14:
Hi Stubbs

I assume that you have to many '{{' around `"{{{{ value_json.temperature }}}}"`.

## Suggestion A

Before passing the strings to `mqtt.publish()`, you could verify it using this code:
```python
json_string =f'{{ "name":  "Gas resistance"...'
print(json.loads(json_string))
mqtt.publish(..., json_string)
```
Looking at the print output, you will quickly see, which variables have not been expanded.

## Suggestion B

Write a python dictionary and let `json.dumps()` do the job for you.
```python
dict_json = { "name":  "Gas resistance",   "state_topic": "..."}
mqtt.publish(..., json.dumps(dict_json))
```

rnlgreen, 2024-06-08:
For info, I just stumbled across this same problem. I had working code until I tried to publish a temperature and included the ° (degree) symbol. 
I haven't the patience to figure out how to solve this, so I've just dropped that symbol!

gurka, 2024-11-28:
The problem is that `umqtt.simple.MQTTClient` assumes that `msg` contains only ascii characters (i.e. that `len(msg)` is the number of bytes that will be written on the socket): https://github.com/micropython/micropython-lib/blob/e4cf09527bce7569f5db742cf6ae9db68d50c6a9/micropython/umqtt.simple/umqtt/simple.py#L120

You can fix this in user code by convert the `str` to `bytes` before calling `publish`, e.g.

```
msg = "Hejsan Världen"
client.publish(topic, bytes(msg, 'utf-8'))
```

I think this is described in the README as well

projectgus, 2025-08-26:
>I think this is described in the README as well

It is indeed. I think the current behaviour of silently truncating utf-8 arguments is not a good choice though - it should either explicitly fail, or attempt to encode even if this does cost some RAM (and the documentation can recommend passing bytes to save RAM).

kdpuvvadi, 2025-08-26:
Directly using ascii characters works.

```python
entity_temparature = Thermostat(
    identifiers="thermo_bedroom",
    manufacturer="Raspberry Pi Foundation",
    model="PicoW",
    device_name="Bedroom Thermostat",
    object_id="thermo_bedroom_temperature",
    unique_id="thermo_bedroom_temperature",
    name="Bedroom Temperature",
    state_topic="home/bedroom/climate",
    value_template="{{ value_json.Temperature }}",
    device_class="temperature",
    unit_of_measurement= chr(176) + "C"
)
```