micropython/micropython-lib #757: ImportError: cannot import name 'MQTTModule' from 'bell.avr.mqtt.client'

victoriadekalita, 2023-10-25:
I keep getting this error, I don't really know what I'm doing this is my team's first year doing AVR drones and if anyone could help that would be awesome. We have done FTC in the past using Visual Studio so we know a little bit about that but we have never done Python in Visual Studio. I cannot figure out for the life of me how to get this stuff to work and would love some tips, please! We got the drone to fly and everything we just need an autonomous and I am struggling to get it to work...

jonnor, 2024-10-12:
Have you installed a module with the path `bell.avr.mqtt.client` ? I am not aware of any such module for MicroPython, certainly there is none provided in the micropython-lib repository (where this issue is filed).

projectgus, 2024-11-19:
Closing as agree this seems unrelated to micropython-lib. @victoriadekalita if you're still struggling with this then feel free to post a question on https://github.com/orgs/micropython/discussions with some more context.

closed 2024-11-19 by projectgus