micropython/micropython-lib #760: aioble: AttributeError: 'module' object has no attribute 'Service'  (and - no instructions, and lack of useful examples)

gitcnd, 2023-10-29:
It is not clear from this page:-

https://github.com/micropython/micropython-lib/tree/master/micropython/bluetooth/aioble#passive-scan-for-nearby-devices-for-5-seconds-observer

how to use any of that code

```
>>> import l2cap_file_server
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "l2cap_file_server.py", line 53, in <module>
AttributeError: 'module' object has no attribute 'Service'
>>> 
```

we need to be told what to do to make it work, and there needs to be working examples to show the features - especially the "scan" - and the examples need to tell you what you need to to do run them.

even the multitests don't seem to work?  (and there's no tests for the scan feature):

```
MicroPython v1.21.0 on 2023-10-06; Generic ESP32 module with OTA with ESP32

Type "help()" for more information.

>>> import notify
>>> notify.instance1()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "notify.py", line 148, in instance1
AttributeError: 'module' object has no attribute 'stop'
>>> 
```

disclaimer: I'm **not** an expert in MicroPython

fpetzold, 2024-07-09:
I had the problem when I try to use aioble compiled in as a module in MicroPython. When I install aioble with mpi on the board itself, everything works.

jonnor, 2024-10-12:
@gitcnd  Does installing the module with mip solve the problem, as suggested by @fpetzold ?

GPwebdesign, 2025-05-25:
Installing the module with mip solve the problem also for me.
thank's @fpetzold