micropython/micropython-lib #761: AttributeError: 'bytes' object has no attribute 'translate'

magixyu, 2023-11-01:
When use the below code, the error apears:
base64.b32decode(activation_code, True, "I")

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onepass.py", line 90, in get_next_code
  File "onepass.py", line 57, in _get_entropy
  File "base64.py", line 269, in b32decode
AttributeError: 'bytes' object has no attribute 'translate'

it looks that the base64 module in micropython-lib is not compatible. Please help to take a look of this issue.

Thanks,
Tristan

jimmo, 2023-11-10:
Thanks @magixyu -- I think this was inherited from the original version of this repo and clearly this code has just been copied from CPython. We should probably remove it until it can be re-implemented.

On most boards you should be able to use the built-in `binascii` for basic base64 encode/decode via `binascii.b2a_base64` and `binascii.a2b_base64`.

magixyu, 2023-11-12:
@jimmo 
Thank you very much for your information and suggestion. Actually I'd like to use the base32 encoding in Base64 module. 
Is there a base32 encoding library in the embed library?

Thanks,
Tristan

magixyu, 2023-11-12:
@jimmo  I created a pull request just for base64 module.

magixyu, 2023-11-12:
Tested on esp32s3. The module base64 works well.

magixyu, 2023-11-12:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1008
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb5c
load:0x403cc700,len:0x2e6c
entry 0x403c989c
MicroPython v1.21.0-1-gf07c51551-dirty on 2023-11-12; ESP32S3
Type "help()" for more information.
>>> import base64
>>> base64.b64encode(b"aaa")
b'YWFh'
>>> base64.b32encode(b"aaa")
b'MFQWC==='
>>> 
```

closed 2023-11-28 by magixyu