micropython/micropython-lib #763: RFC: cleanup of sdcard.py code

mendenm, 2023-11-02:
The sdcard.py module works, but seems a bit incomplete and slow.  After quite a bit of discussion,
https://github.com/orgs/micropython/discussions/12687
I have prepared a prototype of a newer scheme for it.  It is much faster, since it uses an 'adaptive' timeout scheme zero delays unless the a bunch of fast delay loops have passed without data.  It correctly implements crc7 on all commands, and allows the use of crc16 on data.

I have forked the repo, and am preparing a pull request.  Before I go there, I would appreciate any commentary from the PTB about the changes.  I am attaching a zip with the current status.

Thanks for any comments.

[sdcard_20231102_0752.zip](https://github.com/micropython/micropython-lib/files/13238451/sdcard_20231102_0752.zip)

mendenm, 2023-11-04:
Here is a newer version to look at, if anyone is looking and testing.  I have removed some redundant/dead code in the init_v1 and init_v2, so they are a single, inline block.  This is approaching functional final, since I don't see anything else to trim.
[sdcard.20231104_0804.zip](https://github.com/micropython/micropython-lib/files/13256586/sdcard.20231104_0804.zip)

jimmo, 2023-11-10:
This sounds good @mendenm ! I haven't followed the discussion thread in detail, but I get the impression that this should address many of the common issues that people run into with this library. Looking forward to seeing the PR.