micropython/micropython-lib #770: micropython/lora RSSI readings

davefes, 2023-11-20:
For a Hope Radio RF96 (RFM96W) the RSSI readings seem to calculated incorrectly.  I generally see about -110dBm where errors start creeping-in.  Two other Micropython LoRa implementations report -120 or thereabout dBm for similar "errors*.

Is the formulae:
```py
# units: dBm
rx_packet.rssi = self._reg_read(_REG_PKT_RSSI_VALUE) - (157 if self._pa_boost else 164)
```
valid for this chip?  I do not understand why `pa_boost` is in this formulae.

projectgus, 2023-11-21:
Hi @davefes,

Thanks for noticing this, it's a bug. It *should* be based on this section of the SX1276 datasheet:

[image]

Meaning the implementation should be something like:

```py
# units: dBm
rx_packet.rssi = self._reg_read(_REG_PKT_RSSI_VALUE) - (157 if self._rf_freq_hz > 525_000_000 else 164)
```

Are you in a position to test this version and submit a PR to change it? If not, I'll get to it as soon as I have time.

davefes, 2023-11-21:
Not until the weekend.  Submit a PR ... it would be a first for me.  As long as it works it's way into library I am OK with the "under-reporting".

projectgus, 2023-11-21:
Thanks @davefes! Will leave this issue open until there's a fix.

> Another odd behaviour noted: the first RSSI after a re-boot is 5dB better then subsequent RSSI readings.

Interesting. I assume that's probably an artifact of the module, although I don't know what it would be down to.

davefes, 2023-11-23:
Change VALUE to VAL
`
# units: dBm
rx_packet.rssi = self._reg_read(_REG_PKT_RSSI_VAL) - (157 if self._rf_freq_hz > 525_000_000 else 164)
`
and RSSI values around -117dBm are when errors started appearing.  Still see 4 or 5dB higher on the first RSSI reading.  I am sure I haven't seen this happen on other implementation, but go back and check.

davefes, 2023-11-30:
Confirm that RSSI is the same on the first loop and subsequent loops with your chips ... then I'll just say it is problem with Hope Radio RF96 chips and I will close this issue.