micropython/micropython-lib #771: json lib module, __init__.py syntax error

permaBox, 2023-11-21:
The json lib module seems to have an issue. That was not observed in 2018.

To reproduce the error use:
r = requests.get('https://aa.usno.navy.mil/api/rstt/oneday?date=2023-11-21 &coords=48.441832,9.890139 &tz=2').json()

Generated error message:
File "requests/__init__.py", line 33, in json
ValueError: syntax error in JSON

ricksorensen, 2023-11-22:
If I remove the space between date and `&coords` and remove the space abetween coordinates and `&tz` it works:
```
>>> r=requests.get('https://aa.usno.navy.mil/api/rstt/oneday?date=2023-11-21&coords=48.441832,9.890139&tz=2')
>>> r.json()
{'apiversion': '4.0.1', 'properties': {'data': {'year': 2023, 'month': 11, 'fracillum': '61%', 'closestphase': {'phase': 'First Quarter', 'month': 11, 'year': 2023, 'time': '12:50', 'day': 20}, 'isdst': False, 'day': 21, 'day_of_week': 'Tuesday', 'label': None, 'sundata': [{'phen': 'Begin Civil Twilight', 'time': '08:02'}, {'phen': 'Rise', 'time': '08:37'}, {'phen': 'Upper Transit', 'time': '13:06'}, {'phen': 'Set', 'time': '17:35'}, {'phen': 'End Civil Twilight', 'time': '18:10'}], 'tz': 2.0, 'moondata': [{'phen': 'Set', 'time': '00:52'}, {'phen': 'Rise', 'time': '15:09'}, {'phen': 'Upper Transit', 'time': '20:34'}], 'curphase': 'Waxing Gibbous'}}, 'type': 'Feature', 'geometry': {'coordinates': [9.890139, 48.44183], 'type': 'Point'}}
>>> 
```

permaBox, 2023-11-23:
It works ... sometimes, means it is not stable. 
I often get this error:
...
  File "requests/__init__.py", line 180, in get
  File "requests/__init__.py", line 76, in request
OSError: -202

jonnor, 2024-08-25:
In order to be able to debug the issue, you need to provide examples of valid JSON that the library does not parse.