micropython/micropython-lib #773: mip installs old version of `aioble`

iot49, 2023-11-30:
I am trying to install the latest version of `aioble` with:

```
mpremote mip install --no-mpy aioble   
```

The package is installed, but when I look e.g. at `Characteristic.indicate` in `server.py` the signature is

```
async def indicate(self, connection, timeout_ms=1000)
```

On `github/micropython-lib` the signature is

```
async def indicate(self, connection, data=None, timeout_ms=1000)
```

How do I install the most recent version? Unlike standard `pip`, `mip` does not seem to support version controlled installs, as far as I can tell from the documentation. I have no clue where `mip` gets its files from, if not github (master branch).

iot49, 2023-12-10:
Cleared up (embarrassing error).

closed 2023-12-10 by iot49

jimmo, 2023-12-18:
Sorry this was an issue on our end (the build pipeline had stopped). Thanks for reporting!