micropython/micropython-lib #775: OSError: timeout waiting for v2 card

zhuzhutou, 2023-12-05:
## Title: SD Card Initialization Timeout Issue: Lenovo thinkplus platinum Pro 64GB V30 A3 U3

**Description:**

I'm experiencing an initialization timeout issue with a Lenovo thinkplus platinum Pro 64GB V30 A3 U3 SD card on MicroPython version 1.21.0 on my device. When I attempt to initialize the SD card, I encounter the following error message:

SD card initialization timeout error: Error: Timeout occurred during SD card initialization.

**Steps to reproduce:**

1. Insert the Lenovo thinkplus platinum Pro 64GB V30 A3 U3 SD card into the device's SD card slot.
2. Execute the SD card initialization code in the MicroPython environment.

```python
import os, sdcard, machine

def sdtest():
    spi = machine.SPI(1, baudrate=1000000, sck=machine.Pin(10), mosi=machine.Pin(11), miso=machine.Pin(12))
    spi.init()  # Ensure right baudrate
    sd = sdcard.SDCard(spi, machine.Pin(13, machine.Pin.OUT))  # Compatible with PCB
    vfs = os.VfsFat(sd)
    os.mount(vfs, "/fc")
    print("Filesystem check")
    print(os.listdir("/fc"))

    line = "abcdefghijklmnopqrstuvwxyz\n"
    lines = line * 200  # 5400 chars
    short = "1234567890\n"

    fn = "/fc/rats.txt"
    print()
    print("Multiple block read/write")
    with open(fn, "w") as f:
        n = f.write(lines)
        print(n, "bytes written")
        n = f.write(short)
        print(n, "bytes written")
        n = f.write(lines)
        print(n, "bytes written")

    with open(fn, "r") as f:
        result1 = f.read()
        print(len(result1), "bytes read")

    fn = "/fc/rats1.txt"
    print()
    print("Single block read/write")
    with open(fn, "w") as f:
        n = f.write(short)  # one block
        print(n, "bytes written")

    with open(fn, "r") as f:
        result2 = f.read()
        print(len(result2), "bytes read")

    os.umount("/fc")

    print()
    print("Verifying data read back")
    success = True
    if result1 == "".join((lines, short, lines)):
        print("Large file Pass")
    else:
        print("Large file Fail")
        success = False
    if result2 == short:
        print("Small file Pass")
    else:
        print("Small file Fail")
        success = False
    print()
    print("Tests", "passed" if success else "failed")

sdtest()
The SD card initialization timeout error occurs after executing the above code.
Expected behavior:

I expect the SD card to initialize successfully, allowing me to access files on the SD card.

Attempts made:

I’ve checked the physical connection and confirmed that the SD card is inserted correctly.
I attempted to upgrade MicroPython to the latest version, but the issue still persists.
The memory card is formatted with the FAT32 file system.
I have searched through MicroPython documentation and forums but have not found a similar issue.
Environment:

Device model: Lenovo thinkplus platinum Pro
SD card model: 64GB V30 A3 U3
MicroPython version: 1.21.0
Operating system: (if relevant)
Expected resolution time:

If possible, please address this issue as soon as possible as the SD card is crucial for my project.

Thank you for your kind assistance!

mendenm, 2023-12-06:
Please try the updated SDCard library in my pull request:
https://github.com/micropython/micropython-lib/pull/765
and let  me know if it works with this.

ArJini16, 2024-08-23:
With the new code I first got the error OSError: (116, 'card type', 'v2'), which specifies exactly what to do. And after changing the sd card it finally worked. Thank you so much. <3