micropython/micropython-lib #776: problem with iperf3 on MicroPython

mauriceee31, 2023-12-07:
Hi, 

i am getting the following error when I try to run a iperf3 client on my Raspberry Pi Pico W:
` import iperf3
   client = iperf3.Client()

   Traceback (most recent call last):
   File "<stdin>", line 1, in <module>
   AttributeError: 'module' object has no attribute 'Client'`

in this unofficial documentation it should work. https://iperf3-python.readthedocs.io/en/latest/examples.html#examples

dpgeorge, 2023-12-18:
The MicroPython version of `iperf3` doesn't implement all features of the main/CPython version.

To use it in client mode, use commands like this:
```py
iperf3.client('192.168.1.5')
iperf3.client('192.168.1.5', udp=True, reverse=True)
```

jonnor, 2024-08-25:
The appears to be a fix/workaround provided. Proposing to close this issue.

closed 2024-08-25 by mattytrentini