micropython/micropython-lib #781: aiohttp converts data to string

bixb922, 2024-01-03:
Nice to have aiohttp, since it supports the https protocol with 1.22!! I did not see aiohttp in the 1.22 announcement, it really is useful to have async https now!

When posting data with the data= option, the data gets converted to string before sending. This encloses the data in b'' before sending. For example data=b'hello' will send str(b'hello'), i.e. "b'hello'"

What is expected: I expected data= to accept bytes or bytearray, just like urequests or CPython aiohttp, and to transmit the raw bytes without converting nor encoding/decoding. 

I am using: MicroPython v1.22.0 on 2023-12-27; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Sample code (based on aiohttp examples):
```py
import aiohttp
import asyncio
async def main():
    do_connect()
    async with aiohttp.ClientSession("http://httpbin.org") as session:
        async with session.post("/post",data=b'hello world') as resp:
            assert resp.status == 200
            rpost = await resp.text()
            print(f"POST: {rpost}")

asyncio.run(main())
```
The output is
```
POST: {
  "args": {}, 
  "data": "b'hello world'", 
  "files": {}, 
  "form": {}, 
  "headers": {
    "Content-Length": "14", 
    "Host": "httpbin.org", 
    "User-Agent": "compat", 
    "X-Amzn-Trace-Id": "Root=1-6594aa03-4f076b3a3480ef2659dfc035"
  }, 
  "json": null, 
  "origin": "181.43.38.11", 
  "url": "http://httpbin.org/post"
}
```

Carglglz, 2024-01-03:
@bixb922 I think I may have overlooked binary data treatment, let me know if this #782  fixes it 👍🏼

bixb922, 2024-01-03:
Thank you very much for your fast response. Yes, this now works for me, transmitting binary data!!

PR #782 (2024-02-08): python-ecosys/aiohttp: Fix binary data treatment.
Fixes #781, #784, #788
closed 2024-01-03 by bixb922