micropython/micropython-lib #796: LoRa Driver SX126x does not set SyncWord

oeysteinhansen, 2024-02-03:
The driver has the option to set the SyncWord via its configure() command (also via driver init), in the option "syncword". 
Most of the code is correct but when writing the syncword value to the SX126x device registry, it writes to a wrong registry because of a code issue. 

Package: lora-sx126x
MicroPython: MicroPython v1.22.1 on 2024-01-05; Generic ESP32S3 module with ESP32S3
Board: Heltec LoRa 32 v3

code file: micropython-lib/micropython/lora/lora-sx126x/lora/sx126x.py 
code line: 386 
code current: `self._cmd(">BBH", _CMD_WRITE_REGISTER, _REG_LSYNCRH, syncword)`
code fixed: `self._cmd(">BHH", _CMD_WRITE_REGISTER, _REG_LSYNCRH, syncword)`
code note: Changing the data packing format from `">BBH"` to` ">BHH"` so the full 16bit registry address is used.

I have tested the change with creating a separate function (as i have not been able to edit the library code)
```
def lora_write_syncword(modem, syncword: int = 0x12):
    _CMD_WRITE_REGISTER = const(0x0D)
    _REG_LSYNCRH = const(0x0740)
    if syncword < 0x100:
        syncword = 0x0404 + ((syncword & 0x0F) << 4) + ((syncword & 0xF0) << 8)
    # Write Peripheral Register
    modem._cmd(">BHH", _CMD_WRITE_REGISTER, _REG_LSYNCRH, syncword)
```
Tested by receiving messages from Meshtastic v2 devices witch use the syncword 0x2B.

projectgus, 2024-02-13:
@oeysteinhansen Thanks for the very clear bug report! I've got a fix for this ready now, exactly along the lines you suggested.

oeysteinhansen, 2024-02-15:
Thank you for your quick response. Had a look at the #804 committed code. Looks correct.
I will update when a version with the fixes is available.

PR #804 (2024-02-20): lora: Multiple small fixes (including syncword, sx126x minimum config)
Collection of small LoRa fixes in individual commits. See commit messages for more details.

* SX126x: Fix invalid configuration after reset. Configuring with the documented minimal `lora_cfg` didn't produce the documented results, some configuration keys always needed to be set.
* SX126x: Fix syncword setting. Closes #796.
* SX127x: Fix syncword setting. Implementation was simply missing. :facepalm: 
* lora-sync: Fix a possible race where a send which failed immediately (or completed very quickly) would not return immediat [... cut]
closed 2024-02-20 by dpgeorge