micropython/micropython-lib #797: tarfile-write issue with file/folder permissions

ubidefeo, 2024-02-03:
### Context
I've been playing with `tarfile` for a few days and yesterday started trying to create my own archives using `tarfile-write` installed via `mip`.

Adding a folder to it works recursively, but transferring the file to the PC via `mpremote` and expanding it yields all files and folders with `--- --- ---` permissions (0x000).

### Discovery
I investigated and found the source of the issue is related to how the `TarInfo` object [is composed in these lines](https://github.com/micropython/micropython-lib/blob/7cdf70881519c73667efbc4a61a04d9c1a49babb/python-stdlib/tarfile-write/tarfile/write.py#L102C9-L106C32)

in particular on line 102 we find `mode` to be set to the `os.stat(FILE)[0]`

```
tarinfo.mode = stat[0]
```

`stat[0]` is either `0x8000` (file) or `0x4000` (folder).
And the subsequently applied masking yields `0x000` as permissions set while adding it to the archive.
I think that mode should be safely set to `0x1a4 > 644 > rw- r-- r--`

I have tested by manually composing a TarInfo object to be handed to `addfile()` rather than patching `add()`

### How to reproduce:
```python
import tarfile
import os

os.mkdir('test_folder')
f = open('test_folder/test_file_01.txt', 'w')
f.write('file 01')
f.close()

os.mkdir('test_folder/sub_folder')

f = open('test_folder/sub_folder/test_file_02.txt', 'w')
f.write('file 02')
f.close()

archive = tarfile.TarFile('test_archive.tar', 'w')
archive.add('test_folder')
archive.close()

```

Copy the archive from the board to the PC via `mpremote cp :test_archive.tar test_archive.tar`

Expand the archive and verify the permissions for **test_folder**.
From Mac os I was not able to `chmod 644 test_folder`, but using the UI I could unrestrict access.
[image]

Looking at the folder content you can verify the same permission issue happens with `sub_folder` and contained file(s)
[image]

Environment:
I run a custom build for ESP32-S3, but this issue is not related to changes I have applied
(name='micropython', version=(1, 23, 0, 'preview'), _machine='LilyGo T-QT Pro with ESP32S3', _mpy=10758)

MicroPython v1.23.0-preview.48.g076516d88.dirty on 2024-01-21; LilyGo T-QT Pro with ESP32S3
Board: ESP32-S3
TarFile version: '0.4.1'

Hope someone can take a look :) 
u.

ubidefeo, 2024-02-05:
upon further digging, the `.mode` attribute seems to be doing its thing when it comes to `TarInfo`, and the issue lies into how the Tar header is composed.
Dissecting this line
```python
hdr.mode[:] = b"%07o\0" % (tarinfo.mode & 0o7777)
```
the permissions always come out as `000`

The fix I have tested is to replace that line with 
```
hdr.mode[:] = b"%07o\0" % (0x1ed & 0o7777)
```

This yields tar files which can be expanded on the host computer, but of course all the files will be `755`, which might have some implications if one wants to be picky

ubidefeo, 2024-02-05:
Moreover, the way the initial `mode` is set in the `TarInfo` init is wrong, since it characterises the `mode` based on the presence of a `/` at the end of the resource name.
The correct way to do this is to check for `os.stat(name)[0]`
```
-        self.mode = _S_IFDIR if self.name[-1] == "/" else _S_IFREG
+        self.mode = _S_IFDIR if os.stat(name)[0] == 0x4000 else _S_IFREG
```

projectgus, 2024-02-06:
I left a review on your PR, but I wanted to say thanks for a very comprehensive and clear bug report and analysis!