micropython/micropython-lib #802: requests: requests.get() blocks infinitely

tacker66, 2024-02-08:
I am trying to call request.get(<some REST API call>) on a Pico W but this call blocks infinitely.
Using a timeout or non-blocking mode (timeout=0) doesn't help.
The call works with CPython and in Chrome and returns some small JSON.

When using ```self._cached = self.raw.recv(10000)``` instead of ```self._cached = self.raw.read()```
in ```requests/__init__.py (line 20)``` the call works fine.

So shouldn't this ```read()``` be replaced with a loop which tries to ```recv(<num bytes>)```
until it read less than ```<num bytes>``` and then stops?

dpgeorge, 2024-02-08:
Are you able to provide a URL that shows the issue?  So that someone else can reproduce the problem.

tacker66, 2024-02-09:
This happens when trying to call the [local API of an APSystems EZ1 micro-inverter](https://file.apsystemsema.com:8083/apsystems/apeasypower/resource/APsystems%20EZ1%20Local%20API%20User%20Manual.pdf). Perhaps someone from [Sonnenladen](https://github.com/SonnenladenGmbH/APsystems-EZ1-API) could help here. I raised an [cross-linked issue](https://github.com/SonnenladenGmbH/APsystems-EZ1-API/issues/11) there.

Vacant0mens, 2024-11-03:
I have this same issue calling `requests.get("http://worldtimeapi.org/api/timezone/America/Denver")`
It never times out and just sits there waiting. I can only get it out of that blocking state by hard resetting the device.

> So shouldn't this `read()` be replaced with a loop which tries to `recv(num_bytes)` until it read less than `num_bytes` and then stops?
> 
> Update: Tested this with a different kind of device (calling the ThingSpeak API with `requests.post`) and for that only `read()` works correctly. `recv(num_bytes)` yields no bytes.

From what I can tell, it looks like the underlying `usockets` doesn't allow for a custom timeout arg, and by inheritance, neither does `urequests`. Having a loop in there like @tacker66 suggested would be a good idea, then implementing an optional and/or default timeout time would be fairly easy as well.