micropython/micropython-lib #819: Audio from a microphone

ebolisa, 2024-03-05:
Hi, does anyone know how to interface with ZTS6531S smt microphone? I got working with Circuitpy but I want to use mpython. Any help is appreciated.

jonnor, 2024-04-07:
`ZTS6531S` is a digital microphone with PDM output. Support for PDM will vary based on platform/microcontroller. Which chip are you using?

For ESP32, there is a MR for adding support in https://github.com/micropython/micropython/pull/14176

ebolisa, 2024-04-07:
> `ZTS6531S` is a digital microphone with PDM output. Support for PDM will vary based on platform/microcontroller. Which chip are you using?
> 
> For ESP32, there is a MR for adding support in [micropython/micropython#14176](https://github.com/micropython/micropython/pull/14176)

It's the one which comes with this board [https://52pi.com/products/deskpi-picomate-for-raspberry-pi-pico-pico-w?variant=42514429345944](url). As mentioned before, the microphone module works with the library which comes with CircuitPython OS.
Will look into that MR. Thank you much!

jonnor, 2024-04-07:
That board is for the Raspberry PI Pico / RP2040, not ESP32. There is at the moment no PDM support for RP2040 in MicroPython. There was a MR here with some working code, but it has not been integrated. https://github.com/micropython/micropython/pull/8016

ebolisa, 2024-04-08:
> That board is for the Raspberry PI Pico / RP2040, not ESP32. There is at the moment no PDM support for RP2040 in MicroPython. There was a MR here with some working code, but it has not been integrated. [micropython/micropython#8016](https://github.com/micropython/micropython/pull/8016)

Got you, thank you!

projectgus, 2024-08-27:
Closing in favour of the linked PR. Once PDM support is enabled then it should be possible to make this microphone work.

closed 2024-08-27 by projectgus