micropython/micropython-lib #824: It's "offline" after 10+ seconds, use umqtt.simple

edgexie, 2024-03-11:
My code is like the picture. I try umqtt.simple to connect my raspberry with MQTT. But I found it's offline after 10+ seconds, and then I receive the error code in webrepl. Has anyone had the same problem?
[image]
[image]

felixdoerre, 2024-03-14:
You are using `keepalive=10`. This means that you ask the server to disconnect you, if it does not receive a `c.ping()` in that timeframe. Either set `keepalive=None`, or ensure `c.ping()` is called at least every 10 seconds.

Also, please don't paste code as screenshot, but use backticks.

edgexie, 2024-03-18:
> You are using `keepalive=10`. This means that you ask the server to disconnect you, if it does not receive a `c.ping()` in that timeframe. Either set `keepalive=None`, or ensure `c.ping()` is called at least every 10 seconds.
> 
> Also, please don't paste code as screenshot, but use backticks.

HI , thx for your help, I change the screen shot code to snippets code.

In your help, you mean I need to code for the ping by myself? Doesn't MQTT automatically send pings?

felixdoerre, 2024-03-18:
Yes, that's exactly what I mean. The micropython mqtt-client will not send ping packets automatically. The `keepalive`-parameter just configures what you want the server to expect. You have to call `c.ping()` yourself.

edgexie, 2024-03-19:
I hope a offical demo use `keepalive-parameter` 🤕

closed 2024-03-19 by edgexie