micropython/micropython-lib #828: how to use ssl in umqtt.simple v1.4.0

edgexie, 2024-03-19:
Is there a demo to show how to use ssl in umqtt.simple v1.4.0

Str34kk, 2024-03-24:
the procedure is intuitively simple. To initialize an SSL context, employ the following code:
```
import ssl
ssl_context = ssl.SSLContext(ssl.PROTOCOL_TLS_CLIENT)
ssl_context.verify_mode = ssl.CERT_NONE
```

If your requirement involves utilizing SSL certificates that are not self-signed, you can direct the context to verify the certificate's location with:
`ssl_context.load_verify_locations("path_to_cert")`

Following this configuration, incorporate the SSL context as a parameter when establishing your client connection. This can be achieved as shown below:

```
client = MQTTClient(client_id=clientid, server=hostname, port=port_no, user=user_name, password=passw, keepalive=3600, ssl=ssl_context)
client.connect()
```

uiolee, 2024-04-01:
modified from https://docs.emqx.com/en/cloud/latest/connect_to_deployments/esp32_with_micropython.html#connect-with-ssl-tls

```python
with open("ca.crt", "rb") as f:
        cadata = f.read()
    ssl_params = dict()
    ssl_params["cert_reqs"] = ssl.CERT_REQUIRED
    ssl_params["cadata"] = cadata
    ssl_params["server_hostname"] = SERVER
    sslctx = ssl.SSLContext(ssl_params)
    client = MQTTClient(CLIENT_ID, SERVER, PORT, USERNAME, PASSWORD, ssl=sslctx)
    client.connect()
```

edgexie, 2024-04-08:
> the procedure is intuitively simple. To initialize an SSL context, employ the following code:
>
> ```
> import ssl
> ssl_context = ssl.SSLContext(ssl.PROTOCOL_TLS_CLIENT)
> ssl_context.verify_mode = [...]

HI, I test your code, but the REPL tell me the ssl  'module' object has no attribute 'CERT_NONE'.

edgexie, 2024-04-08:
> modified from https://docs.emqx.com/en/cloud/latest/connect_to_deployments/esp32_with_micropython.html#connect-with-ssl-tls
>
> ```python
>     with open("ca.crt", "rb") as f:
>         cadata = [...]

HI, when I run code `ssl.CERT_REQUIRED`, I get the ERROR
```
'module' object has no attribute 'CERT_REQUIRED'   
```

uiolee, 2024-06-26:
Build from the latest source code.

closed 2024-11-13 by edgexie