micropython/micropython-lib #841: [Aioble] Update Parameters request from Central not answered by Peripheral.

Quaditz, 2024-04-07:
Hello,

see 

[https://github.com/hbldh/bleak/issues/1262](https://github.com/hbldh/bleak/issues/1262)

On Windows 10, no "Sent LE Connection Update" is sent by the Central GATT Client. In that case, everything works fine.

On Windows 11, a "Sent LE Connection Update" is sent by the Central GATT Client. In that case, the connection breaks down.

The link above has Wireshark Bluetooth logs attached from the computer side, which uses Bleak to create a Central GATT Client. I run Aioble on the Peripheral device side as Peripheral GATT Server.

The problem in the link above is, that Bleak crashes in Windows 11 while it works perfectly fine in Windows 10. Our suspicion is, that the Peripheral does not handle the "Sent LE Connection Update" correctly.

On the Aioble side, I also get an error (in the time where the Bleak client freezes on Windows 11):

Traceback (most recent call last):
  File "<stdin>", line 167, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 164, in main
  File "asyncio/funcs.py", line 1, in gather
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 88, in advertise
  File "aioble/device.py", line 295, in __aexit__
  File "aioble/device.py", line 216, in disconnect
  File "aioble/device.py", line 232, in disconnected
  File "aioble/device.py", line 232, in disconnected
  File "asyncio/core.py", line 1, in run_until_complete
  File "aioble/device.py", line 198, in device_task
  File "asyncio/event.py", line 1, in wait
CancelledError: 

The error does not even referr to a line of code that I have written, so it must happen during the 

await connection.disconnected()

stage after I have initiated the connection via

async with await aioble.advertise(...) as connection:
    ...
    await connection.disconnected() 

The same code (on both sides, same server code and same client code) runs perfectly on Windows 10. And the only difference that we see in the Wireshark logs (see link above) is that under Windows 11 a Sent LE Connection Update is sent by the client.

dlech, 2024-04-07:
To be more clear, apparently Windows 11 now requires peripherals to support the "Connection Parameter Update procedure" (e.g. Bluetooth core spec 5.3, vol 3, part C, sec 9.3.9) and is hanging because it never gets a response from an RPi Pico running MicroPtython.

This is a low level procedure that should be handled by BTStack already. So if it isn't I'm guessing there is an issue with pumping BTStack events in the MicroPython code somewhere or a bug in BTStack where it doesn't respond to this request under specific conditions.

In any case, this is a core MicroPython issue, not micropython-lib.

Quaditz, 2024-04-08:
Duplicated, issue tracked via

[https://github.com/micropython/micropython/issues/14268](https://github.com/micropython/micropython/issues/14268)

closed 2024-04-08 by Quaditz