micropython/micropython-lib #844: urequests still using HTTP/1.0

linusoft, 2024-04-11:
The urequests module is still using HTTP/1.0 for HTTP requests:

line 39 in micropython-lib/micropython/urllib.urequest/urllib/urequest.py:

 ` s.write(b" HTTP/1.0\r\nHost: ")`

Since a host header is included, can this be updated to use HTTP/1.1 or even HTTP/2.0?

Some sites (e.g. mongodb) are now responding with '426 Update Required' for HTTP/1.0 requests.

Thanks

tharuka-pavith, 2024-05-05:
Hey! I also think it is better to update the header in  [line 39](https://github.com/micropython/micropython-lib/blob/583bc0da70049f3b200d03e919321ac8dbeb2eb8/micropython/urllib.urequest/urllib/urequest.py#L39C9-L39C40) of `micropython-lib/micropython/urllib.urequest/urllib/urequest.py` as,
```diff
- s.write(b" HTTP/1.0\r\nHost: ")
+ s.write(b" HTTP/2.0\r\nHost: ") 
```
As I know, most major web servers and CDNs provide backward compatibility. Therefore there will be no harm by updating it.

sosi-deadeye, 2024-05-05:
[HTTP/2](https://http2.github.io/) is a binary protocol and uses [HPACK](https://httpwg.org/specs/rfc7541.html) for header compression. Changing this, require additional code to handle the protocol and header compression.

I also think the change to HTTP/1.1 may require some additional code, to handle the different connection types : `close`, `keep-alive`.

In addition, there are different [Transfer-Endocings](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Transfer-Encoding).

mattytrentini, 2024-05-06:
It would be good to merge _back_ some of the changes from the CircuitPython requests library (they've updated it since originally starting with the MicroPython implementation). In it, they respond with v1.1, though I'm not sure it 100% conforms to spec...

https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/main/adafruit_requests.py