micropython/micropython-lib #855: BLE: Bluetooth Low Energy

kerby1998, 2024-05-14:
Hi, I want to know if it this module is compatible with Bluetooth Low Energy.

kerby1998, 2024-05-16:
Ok, because  I've tried to pair my gopro to my raspberry pi pico w with aioble but it didn't work. Some sources on internet state that micropython cannot pair with raspberry pi pico w. Is it true ?

brianreinhold, 2024-05-30:
That is because there is no pairing support for the PICO_W which uses the btstack. You will need to do quite a bit of work to get pairing to work using aioble with the btstack. Code needs to be added primarily to modbluetooth_btstack.c in the Micropython build distribution. We have been working on it for quite a while and still have a ways to go.

kerby1998, 2024-06-02:
ok thanks

closed 2024-08-25 by robert-hh