micropython/micropython-lib #870: SX1262 LORA driver error: RuntimeError: Internal radio Status (2, 1) OpError 0x20

gampam2000, 2024-05-30:
Hi, 

I use a LILYGO T3-S3 with an ESP32-S3 and SX1262 LoRa chip.
Lora send and receive works perfectly fine with this driver: https://github.com/ehong-tl/micropySX126X

I know wanted to test the [async driver](https://github.com/micropython/micropython-lib/tree/master/micropython/lora) of this micropython-lib, but I can't get it to work.

Here is my complete stripped down test program to test both drivers:

```python
from machine import Pin,SPI
import asyncio

import ubinascii

sx = None
#other driver from https://github.com/ehong-tl/micropySX126X
def configure_sync_lora():
    global sx
    from sx1262 import SX1262

    sx = SX1262(spi_bus=1, clk=5, mosi=6, miso=3, cs=7, irq=33, rst=8, gpio=34)
    sx.begin(freq=869.525000, bw=250.0, sf=7, cr=5, syncWord=0x12,
        power=0, currentLimit=60.0, preambleLength=8,
        implicit=False, implicitLen=0xFF,
        crcOn=True, txIq=False, rxIq=False,
        tcxoVoltage=1.8, useRegulatorLDO=False, blocking=False)

# Initialize LoRa Modem
def get_async_modem():
    from lora import AsyncSX1262
    
    lora_cfg = {
        "freq_khz": 869525,
        "sf": 7,
        "bw": "250",  # kHz
        "coding_rate": 5,
        "preamble_len": 8,
        "output_power": 0,
        "implicit_header": False,
        "crc_en": True,
        "syncword" : 0x12,
        "dio3_tcxo_millivolts" : 1800
     }
    
    lora_spi = SPI(1, baudrate=2000_000, miso=Pin(3), mosi=Pin(6), sck=Pin(5))

    return AsyncSX1262(
         spi=lora_spi,
         cs=Pin(7),
         busy=Pin(34),
         dio1=Pin(33),
         reset=Pin(8),
         lora_cfg=lora_cfg,
    )

async def send_coro(modem):
    counter = 0
    while True:
        print("Sending...")
        await modem.send(f"Hello world from async MicroPython #{counter}".encode())
        print("Sent!")
        await asyncio.sleep(5)
        counter += 1

#uses working driver
def send(payload):
    global sx
    print("LoRa Sender")
    sends = ubinascii.hexlify(payload)
    print("TX: {}".format((sends[0:30])))
    sx.send(payload)

print('Start')

async def main_task():

    #comment out the following 4 lines to use alternate driver
    modem = get_async_modem()
    modem.calibrate()
    print('Send')
    asyncio.create_task(send_coro(modem))
   
    #uncomment to run alternate driver
    #configure_sync_lora()
    #send("Test Lora Payload")
    
    asyncio.sleep(5)

try:
    loop = asyncio.get_event_loop()
    loop.create_task(main_task())
    loop.run_forever()
    print("i should never get printed")
finally:
    asyncio.new_event_loop()  
```

The other driver works perfect, with the driver of micropython-lib I get:
```
Start
Send
Sending...
Task exception wasn't retrieved
future: <Task> coro= <generator object 'send_coro' at 3c1622a0>
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 54, in send_coro
  File "/lib/lora/async_modem/async_modem.py", line 65, in send
  File "/lib/lora/sx126x.py", line 616, in prepare_send
  File "/lib/lora/sx126x.py", line 332, in _check_error
RuntimeError: Internal radio Status (2, 1) OpError 0x20
```

Any suggestions, what this means? or how i can fix this?

majje, 2024-06-06:
I had similar problem with the synchronous SX1262 on a Heltec v3, but solved it with the argument dio3_tcxo_millivolts=1800. (My pins are different)

```
return SX1262(
  spi=lora_spi,
  cs=Pin(8),
  busy=Pin(13),
  dio1=Pin(14),
  reset=Pin(12),
  dio3_tcxo_millivolts=1800,
  lora_cfg=lora_cfg,
)
```

gampam2000, 2024-06-09:
Hi, thanks for the reply that solved it!!! Some parameters are not set from the config but directly in the constructor.

here the updated code:
```

 Initialize LoRa Modem
def get_async_modem():
    from lora import AsyncSX1262
    
    lora_cfg = {
        "freq_khz": 869525,
        "sf": 7,
        "bw": "250",  # kHz
        "coding_rate": 5,
        "preamble_len": 8,
        "output_power": 0,
        "implicit_header": False,
        "crc_en": True,
        "syncword" : 0x12
     }
    
    lora_spi = SPI(1, baudrate=2000_000, miso=Pin(3), mosi=Pin(6), sck=Pin(5))

    return AsyncSX1262(
         spi=lora_spi,
         cs=Pin(7),
         busy=Pin(34),
         dio1=Pin(33),
         reset=Pin(8),
         dio3_tcxo_millivolts=1800,
         lora_cfg=lora_cfg,
    )
```

closed 2024-06-09 by gampam2000