micropython/micropython-lib #882: Cannot init USB devices in boot.py

mischif, 2024-06-13:
I tested this on a pico running MP 1.23 and the latest release of usb-device-cdc.

boot.py:
```python
import usb.device
from usb.device.cdc import CDCInterface

IFACE_TWO = CDCInterface()
IFACE_TWO.init(timeout=0)
#usb.device.get().init(IFACE_TWO, builtin_driver=True)
```

main.py:
```python
from time import sleep_ms

def main():
	while True:
		print(IFACE_TWO)
		sleep_ms(1000)

if __name__ == "__main__": main()
```

Running this as-is, `IFACE_TWO` does show up in main.py, but uncommenting the init function causes the process to error out. [This discord post](https://discord.com/channels/574275045187125269/1237556755282792519/1237564628360237158) suggests you can call `usb.device.get.init()` in boot.py, is this no longer true?

projectgus, 2024-06-19:
> causes the process to error out

Please give some more details about this means: what exactly do you do, and what error output do you see?

A soft reset (i.e. Ctrl-D in the REPL) will still disconnect the USB port each time, because the USB interface objects have to all be freed and recreated.

A hard reset (i.e. plugging the rp2 in, or calling `machine.reset()`) should come up immediately with the two CDC ports.

The only difference between boot.py and main.py is that the first time the device boots from a hard reset, it runs boot.py before it initialises USB. If you put the code in main.py instead of boot.py then the rp2 would first try create a device using the default USB interface, then immediately switch to the custom USB interface. This can show up on the host as connect/disconnect/connect cycle, or as an error depending on the timing.

mischif, 2024-06-19:
With the init line commented out in boot.py:
- plug in pico
- `mpremote fs cp boot.py :`
- unplug/replug
- `mpremote run main.py`

I see the print statements I expect:
```
9600/8N1 rts=False dtr=False
9600/8N1 rts=False dtr=False
[...]
```

With the init line uncommented in boot.py:
- plug in pico
- `mpremote fs cp boot.py :`
- unplug/replug
- `mpremote run main.py`

I do not see the print statements I expect:
```
Traceback (most recent call last):
  [...]
  File "/home/mischif/.local/lib/python3.10/site-packages/serial/serialposix.py", line 549, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 5] Input/output error
```

I get dropped back to the terminal, and connecting with `mpremote a0` yields no output.

projectgus, 2024-06-19:
Ah snap, `mpremote run` does a soft reset before it runs the provided file. Because boot.py has loaded the custom USB interface, it disconnects the USB port every time.

(If you copy both boot.py and main.py to the filesystem then main.py works and you can use `mpremote a0` to view the output, but you still can't use `mpremote run` any more.)

I think the fix for this will be the planned work to have mpremote detect when the USB port is in a disconnect/reconnect cycle, and recover from it instead of erroring out.

mischif, 2024-07-10:
Is there a timeline for when the planned work might be finished?

projectgus, 2024-07-11:
Not at the moment, I'm afraid.

ziesemer, 2024-11-17:
I assume this is impacting `mpremote connect` as well (also including a soft reset)?  Running similar as the OP above, I'm finding that I can connect directly to REPL by connecting to its COM port - but not using `mpremote connect`.