micropython/micropython-lib #904: Can someone add a mini version of web3.py library to execute some blockchain transactions

harshpanzer, 2024-07-26:


jonnor, 2024-08-25:
Hi @harshpanzer. Modules can be developed by anyone in the community. If you do not see a module for your needs, then I recommend trying to develop/port it yourself.

The micropython-lib repository is primarily about core and very common Python modules, like those in the CPython standard library, or very widely used packages (that are also relevant for embedded/microcontrollers). At this time, it is unlikely that web3 or blockchain libraries will be included.

abuvanth, 2024-08-25:
check this out for ed25519 library  - https://github.com/cyberfly-io/cyberfly-micropython/blob/main/crypto_c_lib/src/oryx_crypto.c

example usage - https://github.com/cyberfly-io/cyberfly-micropython/blob/main/cyberfly_sdk/pact.py