micropython/micropython-lib #921: USB HID keyboard and mouse

aloneguid, 2024-09-09:
Does anyone know how to create a USB device that acts as a mouse and keyboard at the same time? Examples here https://github.com/micropython/micropython-lib/tree/fbf7e120c6830d8d04097309e715bcab63dcca67/micropython/usb/examples/device are excellent, but I can only manage to work exclusively as a mouse or a keyboard, but not both at the same time.

As long as I understand, I need to prepare a report descriptor in a special way, but I'm not a HID expect and not sure how to do it. Please share if you have done it before ;)

dov, 2025-06-11:
Before finding this issue, I just asked the same question in micopython discussions:

https://github.com/orgs/micropython/discussions/17475

I also copied my question into github copilot and the response that I got was more or less "it is currently not supported", and it offered to help writing a combined mouse and keyboard hid module.

aloneguid, 2025-06-13:
It's beeen a while! Since then I have done it multiple times. It is possible, but you will have to write your own device rather than use `KeyboardInterface` and `MouseInterface`. It's not possible to run two at the same time, because those would be two separate physical devices, so you would need two physical USB ports, which is also not supported. I should write a blog post, just didn't get to it.

In general, I had to write my own device anyway, because I wanted to add support for extra mouse button and a wheel, which is not part of `MouseInterface`. Currently I have a device which works as a keyboard, mouse, joystick and consumer control at the same time. I'll try to detail this next week.

A bit off topic regarding the discussion - I was trying to do the same a while ago - wireless control via wifi or bluetooth. I settled on bluetooth at the end, because it can work "offline". But I found that even keyboard support wirelessly is a bit of a challenge with MicroPython due to performance issues. There is a lot of code regarding throttling and aggregating keyboard events to make keyboard type close to a normal bluetooth keyboard. Moving a mouse wirelessly is not possible at all due to massive latency issues. Good luck though!

dov, 2025-06-15:
Hi @aloneguid . Thanks for your reply. I'd love to get a copy of your combined keyboard and mouse usb driver! Was it in pure python, or did you need to change the micropython C-Code?

Regarding latency, I'm aware that it is going to be bad, but I'd still like to experience it on my own. I did a simple ping latency of a web sockets server running on the MCU, and it was between 6-300ms. The worst case is clearly unacceptable, but it happened only occasionally.

In any case my idea was to be able to remote control another computer by, and since it is not going to be continuously, perhaps the latency isn't too annoying.