micropython/micropython-lib #923: bluetooth feature workable or not?

kean0048, 2024-09-21:


kean0048, 2024-09-21:
Hi,

I notice that there are some files about ble under micropython/bluetooth, does it workable for RTL872x modules for micropython?

jonnor, 2024-10-12:
Which port are you using for the RTL8722? Questions about what the port supports or not would be best directed to the developers of that port.

The aioble high-level BLE package should work if the port provides a compatible low-level "bluetooth" module.

dpgeorge, 2024-10-18:
The RTL8722 is not supported by MicroPython.

closed 2024-10-18 by dpgeorge

andrewleech, 2024-10-26:
The drivers you're pointing at are for Nordic semi chips, not Realtek. There is no support for any Realtek chips in this official micropython repository.

Realtek have started work on adding support for micropython on their chips here: https://github.com/ambiot/ambd_micropython

Their work is unfinished and/or hasn't been submitted for inclusion in official micropython yet.

As such no support can be given for these chips here, but you could raise a discussion / issue on their own repositories to see if they have Bluetooth support yet.