micropython/micropython-lib #928: shutdown function of logging module does not empty _loggers

xlla, 2024-10-07:
if I want to reuse logger after call shutdown(), I can recreate log_a ,  add file_handler ,  but it will failure on log.xxx

<pre>
Traceback (most recent call last):
...
  File "logging.py", line 141, in info
  File "logging.py", line 135, in log
  File "logging.py", line 71, in emit
ValueError: 
</pre> 

the root cause is, after execute logging.shutdown() , all logger remain in _loggers variables, so all handlers remain in logger.handlers, but the old file_handlers was closed, so emit method failure.

we can run follow code to verify it.

<pre>

import logging
log_a = logging.getLogger('a')
log_a.setLevel(logging.DEBUG)

file_handler = logging.FileHandler('log.txt')
file_handler.setLevel(logging.DEBUG)

formatter = logging.Formatter("%(name)s - %(levelname)s - %(message)s")
file_handler.setFormatter(formatter)
log_a.addHandler(file_handler)

log_a.debug('write to file')

print(logging._loggers)
print(log_a.handlers)
logging.shutdown()

log_a = logging.getLogger('a')
log_a.setLevel(logging.DEBUG)

file_handler = logging.FileHandler('log.txt')
file_handler.setLevel(logging.DEBUG)

formatter = logging.Formatter("%(name)s - %(levelname)s - %(message)s")
file_handler.setFormatter(formatter)
log_a.addHandler(file_handler)

log_a.debug('write to file')

print(logging._loggers)
print(log_a.handlers)

</pre>

to fixed it, pass name to pop method in line 213

`logging._loggers.pop(logger.name, None)`

jamesquilty, 2026-10-17:
I just saw the same thing. An alternate fix which makes use of both values returned by `items()` is:

```diff
def shutdown():
-    for k, logger in _loggers.items():
+    for name, logger in _loggers.items():
         for h in logger.handlers:
             h.close()
-        _loggers.pop(logger, None)
+        _loggers.pop(name, None)
```

Josverl, 2026-10-17:
Could you create a PR with your suggested fix ?

jamesquilty, 2026-10-18:
@Josverl Yes, I'd be happy to! 😄

I've forked the repo and read the CONTRIBUTING document. There are a couple of bugs which might be fixed in the same PR, if that's the typical or preferred way of doing things? They are:

1. `basicConfig()` calls `handler.setLevel(level)` where `level=WARNING`, setting the root logger's handler level to `WARNING`. It shouldn't do this, the CPython module sets handler levels to `NOTSET` by default so that all messages are processed [1]. The consequence of setting the root handler level is that it acts as a global minimum threshold for named loggers which don't define their own handlers and thus inherit/use the root logger's handler. The fix is easy: delete the line which sets the handler level.
2. `CustomFormatter.format` hard-codes a dict of `Logrecord` attributes in its return statement when it should really just `return self.fmt % record.__dict__`. This is more of a refactor 🙂 

There are also a couple of open Issues "logging.FileHandler not adding new lines to log file" and "logging.basicConfig(force=False) on empty handler list never initializes Formatter and Handler" which I could potentially take a look at in this PR too, and some documentation updates?

Please let me know if the all-in-one-PR approach is preferred? Also, I'm used to an Issue-tracker based approach from GitLab, so my first instinct is to raise a separate Issue for (1) above - please let me know if this would be appropriate?

#### References

1. <https://docs.python.org/3/library/logging.html#logging.Handler.setLevel>