micropython/micropython-lib #930: Keyboard Multimedia Key Support?

bai-yi-bai, 2024-10-20:
Hello, these are awesome libraries. I would like to implement USB HID keyboard multimedia keys, but I cannot get them to work. I am using the `keyboard-example.py` and `keyboard.py` file as a starting point. Is there some trick to this?

My project idea is to use a Pico W connected to a PC which doesn't have Ethernet, but does have Wi-Fi. I want to wake it up using usb-wake and then put it back to sleep. The Pico W will connect to an MQTT server and send keyboard commands based on the payload. The main multimedia keys I'd like to send are "Sleep", "Mute", and "Power". The host machine is Debian Bookworm running KDE.

I saw that on a subset of keycodes are defined in [keyboard.py](https://github.com/micropython/micropython-lib/blob/68e3e07bc7ab63931cead3854b2a114e9a084248/micropython/usb/usb-device-keyboard/usb/device/keyboard.py#L118). I was able to add some additional keys, including `KP_DOT = 99 #0x63`, but not every value above 100 worked. I took line 85 of [keyboard-example.py](https://github.com/micropython/micropython-lib/blob/68e3e07bc7ab63931cead3854b2a114e9a084248/micropython/usb/examples/device/keyboard_example.py#L85) and changed it to `keys.append(code)` to `keys.append(my_key)` and set my_key to various integer values. I got the mapping from [this gist](https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2)

I tried several values:

- #my_key = 102 # KEY_POWER 0x66
- #my_key = 248 #define KEY_MEDIA_SLEEP 0xf8
- #my_key = 239 #define KEY_MEDIA_MUTE 0xef
- #my_key = 127 #define KEY_MUTE 0x7f

Unfortunately, none of these values worked, but I confirmed setting `my_key = 4` did print the letter a, and so did many basic input keys and function keys. I plugged in a keyboard which has a mute key and used `xev` to examine the output.

a
```
KeyRelease event, serial 40, synthetic NO, window 0x7600001,
    root 0x1e3, subw 0x0, time 16518608, (89,64), root:(1920,1047),
    state 0x10, keycode 38 (keysym 0x61, a), same_screen YES,
    XLookupString gives 1 bytes: (61) "a"
    XFilterEvent returns: False
```

Mute
```
KeyRelease event, serial 52, synthetic NO, window 0x9000001,
root 0x1e3, subw 0x0, time 15116196, (108,91), root:(1939,1074),
    state 0x10, keycode 121 (keysym 0x1008ff12, XF86AudioMute), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

The mute key on the keyboard does mute system audio, but I cannot get the Pico to mute the PC.
The output states the keycode value should be 121, but when I try that, nothing comes through on xev. Comparing both events, I see that the keysim value is much larger for the mute key versus the a key.
I also don't have a keyboard with a power or sleep button on it, so I cannot verify what those key codes for real hardware are.

Any ideas?

Xenosplitter, 2025-01-26:
I'm running into the same issue, albeit from a different angle. I'm trying to send the keycode for F13-F24, keycodes 0x68-0x73. I think it's related to the report description in [usb-device-keyboard](https://github.com/micropython/micropython-lib/blob/e4cf09527bce7569f5db742cf6ae9db68d50c6a9/micropython/usb/usb-device-keyboard/usb/device/keyboard.py#L108). I'm able to get additional keys (such as F24) working by increasing the usage maximum byte above it's preconfigured 0x65, but multimedia keys still do not work.

I have a hunch it has to do with how the USB HID definition is written, but I haven't figured it out yet. It looks like it was kinda just copied from an [example keyboard definition](https://www.usb.org/sites/default/files/documents/hid1_11.pdf) (Page 69) that likely doesn't cover such use cases.

bai-yi-bai, 2025-01-26:
I'm glad you replied @Xenosplitter, unfortunately I wrote this issue and set the project aside. I also suspect that the amount of data is larger than the library currently accepts, but most of my education comes from Ben Eater videos on the topic. I have also heard about some USB HID keyboards which implement their layer system as multiple virtual keyboards. I'm not sure how far this USB rabbit hole goes and how much is supportable by micropython.

I have a another dream that one day I will be able to replace the cheap mouse and keyboard switch I use on multiple computers with a master and client architecture across multiple RPI pico Ws. Currently, whenever I switch connections, the USB renegotiation takes 3-5 seconds. I've read that more expensive KVMs are able to maintain USB device connections. I'd like to be able to selectively send keyboard or mouse presses to a specific target device (or multiple devices), as well as be able to have a web interface to copy and past text to "type".