micropython/micropython-lib #994: umqtt.simple cannot transmit longer messages than 125 ascii characters on Pico1 W

julian-gpsdk, 2025-04-02:
tried different libraries 

started with https://github.com/peterhinch/micropython-mqtt.git

then https://github.com/bobveringa/micropython-mqtt.git

but now reverted to umqtt.simple and they all exhibit same behavior

gradually increasing message lenght up to 125 characters, when reached wireshark show multiple retransmits and the publishing stalls

tried with 200ms to 1000ms interval, no difference

 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'(
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,-
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,-.
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,-./
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,-./0
this one stalls
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~ !"#$%&'()*+,-./01

tested with a socket tcp transfer and i can transmit 20000 bytes without problem

Josverl, 2025-04-02:
both the MQTT client and server can enforce message limits.
AFAIKT the [mqtt.simple client uses a 2Mb max](https://github.com/micropython/micropython-lib/blob/96e17b65d128cac2328e10f317241d3a64aca2c7/micropython/umqtt.simple/umqtt/simple.py#L123), well over the limit you report. 

What server are you sending to? could it be restricting ?
have you tested using a different client such as https://mqtt-explorer.com/ ?
what port are you using ?

julian-gpsdk, 2025-04-03:
I have exactly tried to send long messages to the mosquitto broker from MQTT explorer without any problems. 

i have tried many things to debug

https://github.com/peterhinch/micropython-mqtt/issues/170

but now i have found that it works with older versions of micropython

it breaks first time with version RPI_PICO_W-20240602-v1.23.0.uf2 and even the last preview build RPI_PICO_W-20250327-v1.25.0-preview.428.g50da085d9.uf2 is also broken

ill stick with the old version for now, im not so experienced in github usage that i know how to file this error in the proper way yet

Josverl, 2025-04-04:
I could not reproduce an error in mqtt.simple wrt to topic or message length 
See [ reponse in other thread](https://github.com/peterhinch/micropython-mqtt/issues/170#issuecomment-2777852839 ) for details.

**I recommend to close this issue** , and re-open it ifwhen the issue can be reproduced.

nastiliano, 2025-04-08:
Hello,

I don't know what the problem is.

But I have the same error with my ESP32.

Just rising the sended data the executions freezes.

Thank you very much.

PD:  https://github.com/orgs/micropython/discussions/17065