micropython/micropython-lib #996: aioble not happy with characteristic writes...

watercoluke, 2025-04-08:
Howdy Legends,

Getting some weird errors when doing the most basic writes...

I am connecting just fine to an ESP32-S3 running Firmware: 3.4.0; MicroPython v1.24.1 on 2024-11-29., and aioble is the latest.

With `capture=True` on the characteristic, when I attempt a write from LightBlue I get:

```
Unhandled exception in IRQ callback handler
Traceback (most recent call last):
  File "/lib/aioble/core.py", line 71, in ble_irq
  File "/lib/aioble/server.py", line 48, in _server_irq
  File "/lib/aioble/server.py", line 184, in _remote_write
AttributeError: type object 'BaseCharacteristic' has no attribute '_capture_queue'
```

...and then with `capture=False` just not getting anything.

Anything obvious I am missing?

Best,
Luke

watercoluke, 2025-06-03:
Here is another clue, again just following the example way to receive a written event.  On startup I'm getting:
```
Registering service...
assert lld.c 422, param 00000000 00000000
Service registered.
```

Anyone ever seen this assertion?  I'm running a plain Espressif ESP32-S3-DevKitC-1 and latest MicroPython release, following the basic aioble examples.  All pretty out of the box stuff, but getting errors :(

Josverl, 2025-06-03:
May I suggest you change this into a [discussion](https://github.com/orgs/micropython/discussions/categories/general), and only raise an issue here when you are sure there is a bug ?
I think you will get more responses there.