micropython/micropython-lib #1013: aioble install fails

leenowell, 2025-05-29:
Hi All,

I am new to micropython and am trying to install the aioble library to my esp8266.  Looking online it seems that I need to connect the board to my local network and then use mip to install it in repl.

I am using the following commands

```
> import mip
> mip.install("aioble")
```

It copies across a few files then I get the following error message

```
>>> mip.install("aioble")
Installing aioble (latest) from https://micropython.org/pi/v2 to /lib
Exists: /lib/aioble/__init__.mpy
Exists: /lib/aioble/core.mpy
Exists: /lib/aioble/device.mpy
Exists: /lib/aioble/peripheral.mpy
Exists: /lib/aioble/server.mpy
Exists: /lib/aioble/central.mpy
Exists: /lib/aioble/client.mpy
Exists: /lib/aioble/l2cap.mpy
Exists: /lib/aioble/security.mpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mip/__init__.py", line 1, in install
  File "mip/__init__.py", line 1, in _install_package
  File "mip/__init__.py", line 1, in _install_json
AttributeError: 'str' object has no attribute 'rpartition'

```

The micropython firmware I downloaded was - ESP8266_GENERIC-20250415-v1.25.0.bin

Is there a bug in the library?

Many thanks in advance

andrewleech, 2025-05-29:
Because the 8266 is resource constrained (compared to many newer devices) the official micropython builds only include the core features; some of the extra features are clearly needed to run mip. You could instead run mip via mpremote when the 8266 is plugged into computer `mpremote mip install <package>`

leenowell, 2025-05-30:
Thanks very much @andrewleech my esp32 arrived and have now managed to get mip working.  The problem now is that when I upload my project (just a main.py) using pymakr, deletes the lib folder on the device :(  I have tried added lib to the py_ignore section of pymakr.conf and setting it in the global settings and no joy.

leenowell, 2025-05-31:
Hi @andrewleech Ordinarily, I develop on my ESPs using the SDK and in C.  I am teaching my daughter to develop and she wanted to do an IoT project so thought I would switch to micropython which would give her some more generic skills.  For this reason I wanted to use a standard IDE (vscode) so came across the pymakr plug in but of course the problem is that this is no longer supported.  It used to have a config setting to not delete libs but having now gone through the pymakr code it looks like that has been depricated.

Your setup sounds like it would work well and be more generic which is good. Would you mind sharing a bit more about your setup.  E.g. how do you get the micropython auto complete/ syntax checking etc. Also to upload/ run the files have you created mpremote scripts or something to automate the connect, upload, reset cycle?  Any help would be appreciated

andrewleech, 2025-06-01:
The vscode type stubs for autocomplete etc come from here: https://micropython-stubs.readthedocs.io/en/main/

I'm more commonly using custom hardware so I have a submodule of micropython itself which I'll custom compile to include features & frozen code into the firmware itself. This is what I'd consider a somewhat more advanced setup though.

For simpler projects I'll have the stubs installed and setup and then use `mpremote mount src` to live load it on the device during development then `mpremote cp -r src/* :` to copy it all in once finished. 

I'll usually wrap these in a simple vscode tasks.json marked as "build" tasks so in vscode I can hit ctrl-b and run one of them. That being said the mpremote commands and running things in a terminal are pretty natural for me now because I do this stuff every day for work.

I have seen a mpremote vscode plugin that I think gives a couple of simple menus for things like this too but haven't tried it myself, I kinda prefer the built in task system.

leenowell, 2025-06-01:
thanks very much @andrewleech much appreciated.  Will give you setup a go and see which one she gets on with better.

leenowell, 2025-06-01:
Thanks very much @Josverl .  I tried MicroPico but couldn't work out how to tell it is an esp32 rather than a pico and also it looked like communication with the device had to be done via wifi.  At least when I click on the MicroPico device controller int the left hand bar (a strawberry looking symbol) there are only wifi devices and asks me to give the device my router credentials.  Also, ideally I would also use it on my esp8266 although that isn't mandatory.  Have you managed to get this working with an ESP32 - e.g. get it to recognise the esp specific commands and upload etc.? Maybe I should have persevered more with it?

closed 2025-07-15 by jonnor